Proto-App/Proto-Android #4: fine tuning

Ashwin-droid, 2021-01-19:
please make that we can fine tune anything. for eg you want to simulate a ac circuit then fine tuning the frequency to 50hz/60hz is a big pain so make a edittext that we can change instead of the stupid dial only. you can keep the dial to coarse tune or set the range like hz,khz,mhz,ghz etc

Proto-App, 2021-01-19:
We have a plan to upgrade dial with **fine tuning mode** so you will be able to set desired value easily.

64-Tesseract, 2021-01-31:
> please make that we can fine tune anything. for eg you want to simulate a **mains ac circuit** then **fine tuning** the frequency to **50hz/60hz** is a **big pain** so make a edittext that we can change instead of the **stupid dial** only. you can **_keep the dial to coarse tune or set the range like hz,khz,mhz,ghz etc_**

I agree with Ashwin - being able to just type values would be faster than moving your finger around a circle.

Proto-App, 2021-04-22:
Feature implemented in version 0.24.0

Tap on dial center to show input dialog
[image]

closed 2021-04-22 by Ashwin-droid