Proto-App/Proto-Android #6: Oscilloscope breaking

Ashwin-droid, 2021-01-22:
I cannot monitor current and voltage. it breaks the program. sol please add current markings to the other side and keep the voltage marking on the left side.

Proto-App, 2021-01-23:
For now it is not possible to show current and voltage signal on one plot. We are working hard on scope addon and there you will be able to show voltage/current with some analysis (RMS, P-P, MAX, MIN, AVG), see early preview:

[image]
[image]

Ashwin-droid, 2021-01-23:
Still no current plot. I am looking forward to having those features. Please don't make it paid

JunkMcNuggets, 2021-12-17:
> 

Unfortunately, they missed that last tiny bit of your request...

Proto-App, 2021-12-17:
@JunkMcNuggets  You can use regular scope ;)

TS232CZ, 2022-07-09:
Is it really possible that average is higher than min and max valueas?
[image]

Proto-App, 2022-07-10:
@TS232CZ 
It looks like bug, could you share your circuit via mail?

TS232CZ, 2022-09-29:
> @TS232CZ It looks like bug, could you share your circuit via mail?

Sent, thank you
TS232

salahchafai, 2022-11-20:
It's been almost two years and it's still impossible to show current and voltage signal on one plot, even with the scope addon! Are there any projections to when we can have this feature?

Proto-App, 2022-11-20:
@salahchafai 
Sorry for big delay but as you can see we develop more and more components in our app. In incoming version 1.14.0 you can expect more dependent sources. Finally in 1.15.0 (january 2023) we will add current and voltage on one plot, things like fit Y axis and more.

Proto-App, 2023-01-21:
Implemented in version 1.15.0

closed 2023-01-21 by Proto-App