Proto-App/Proto-Android #10: Moving components disconnects  wires

Bushmills, 2021-02-01:
Rearranging a circuit by moving components becomes very tedious, when all connections need to be redrawn. Keeping them connected when moving a component may be a huge facilitation when prettifying a circuit. Albeit this would require some sort of autorouting wires, which is admittedly not the most trivial thing to implement.

Proto-App, 2021-02-01:
We are aware that autorouting can greatly improve user experience. Of course we have this improvement on our list, but as you mentioned this topic is quite complex and it requires some time.

Bushmills, 2021-02-03:
There may be a way to solve this in two steps, and have the benefit of being able to move components without automatic routing: keep components connected by "rubber band"-ing their connections in straight lines, no intelligent routing used at all. Albeit far from ideal in situations where multiple connections share the same space, this should still be preferable over the current situation which disconnects the component and leaving the traces behind, as it kept functional integrity, while being comparatively easy to implement. Add more intelligent routing at your convenience and timing - the pressure of needing it to solve the problem with moving components has been alleviated by then already.

Proto-App, 2021-02-04:
Sounds interesting, it should be way easier! Thanks for great idea. We will probably implement this mechanism as a first iteration of "autorouter".

Proto-App, 2021-03-12:
Implemented in v0.21.0

closed 2021-03-14 by Proto-App