Proto-App/Proto-Android #11: Don't just rotate components

Bushmills, 2021-02-01:
But also provide a way to flip them, horizontally and/or vertically.

Proto-App, 2021-02-01:
Firstly we want to bring more components to use. After alpha stage of PROTO app this feature will be our priority.

joaoluiz1977, 2021-02-19:
sites
ja to memorizando de tempo
todos os transistos!
em varias funções em rele, pisca,pisca etc...
quero usar essa nossa memória  em nosso 
aplicativo proto ora evoluir novas ideias
e mais simbologias em LIGAÇÃO PROTO

Proto-App, 2022-01-16:
In version 1.4.0 you can flip transistors in both axis. More components will get this functionality in future.

Bushmills, 2022-01-16:
Thank you, I alraady noticed when I installed the update from Play store. Very useful indeed, makes circuits look much less messy by removing unneeded crossing of wires., and I consider it, albeit mostly a cosmetical matter, a big improvement.