Proto-App/Proto-Android #9: Zooming lack linearity

Bushmills, 2021-02-01:
When zooming in or our, using two finger gesture, no zooming takes place at the begin of gestures, then suddenly zooming sets in way too much rendering it hard to impossible to set zoom level to the desired scale. The displayed circuit seems to be too small or too large, no matter how hard I try, and feels like zoom level changes to integer multiples, without levels between. Device is a 
OnePlus3.

Proto-App, 2021-02-01:
Thanks for feedback, it could be high delta between zoom values in next iterations of UI thread. We will try to optimize this behavior in next versions.

Bushmills, 2021-02-12:
V0.19 is very much an improvement

Proto-App, 2021-02-13:
Yes, behavior should be way better than before.

Proto-App, 2021-02-21:
In case of any issues please reopen this topic.

closed 2021-02-21 by Proto-App