Proto-App/Proto-Android #16: Separate oscilloscope

thelearn-tech, 2021-02-03:
Have a separate oscilloscope will help to probe different components individually and will be even more better if it has multiple channels for example two or three channels

Proto-App, 2021-02-03:
If we understand correctly this is behavior similar to https://github.com/Proto-App/Proto-Android/issues/6

thelearn-tech, 2021-02-04:
Could be I am not too sure .
By the way the best spice simulator ever on Android

Ashwin-droid, 2021-02-05:
> Could be I am not too sure .
> By the way the best spice simulator ever on Android

Yes it is getting there. Very good efforts. Good job, team proto keep it up

HubbsInHobby, 2021-09-19:
I hope the scope can work for both voltages and currents of different components instead of switching between voltage and amperage.

Example: After adding a couple of voltage-scopes on my circuit, I added a current-scope. All the display switched to current and no more voltages is in view!

Proto-App, 2023-01-21:
Implemented in version 1.15.0

closed 2023-01-21 by Proto-App