Proto-App/Proto-Android #18: [Feature request] arcing with high voltage

Ashwin-droid, 2021-02-05:
I think you can create some part called breakout points which if you would provide high enough voltage to estimate the arc length for that voltage and a cool feature is to estimate the size of arc if the current is high enough and also tell the characteristics of the arc like high frequency arcs are hard to blow out than low frequency and high current arcs are also hard to blow out.

Tesla coil of ratio of 1:100 (which is more than enough) could be simulated using transformer in the app.

It could be used to fine tune a tesla coil

And also if you could make that we could set a custom no of turns in the winding of the transformer it could help in finding the resonance frequency of the circuit too.

Ashwin-droid, 2021-02-05:
Ofcourse this is a feature request which I thought was cool to implement but I can understand that you are trying you best to take your app out of alpha.

Proto-App, 2021-03-04:
Sorry for delay, we have some kind of "spark gap" component on our TODO list.

Ashwin-droid, 2021-03-14:
See the last paragraph of setting a custom number of primary and secondary turns and you should also be able to set the distance of the spark gap which could be helpful for making a marx generator

Proto-App, 2021-12-30:
Spark gap is already implemented in app (v1.0.0)

closed 2021-12-30 by Proto-App