Proto-App/Proto-Android #19: Wire joints render the same as wires crossing over

64-Tesseract, 2021-02-22:
In real circuit drawings, a wire passing over another has a bridge-like arc to show that they are separate, and points where more than 2 wires connect have a dot to symbolize a joint. While you _can_ tap on wires to see where their end points are, adding dots and arcs would still make it much easier to see everything.

64-Tesseract, 2021-02-22:
[image]
This is what I mean by "dots" and "arcs", if needed.

Proto-App, 2021-02-22:
Thanks for report, it should improve overall UI experience. We are sure that this should be implemented after alpha stage.

HyVong007, 2021-06-25:
Yeah i also have the same ideas like everyone here. Love PROTO so much (because it is free and strong)

Proto-App, 2021-08-18:
@64-Tesseract 
Wire joints are present in version 0.30.0. We will cover wires crossing later on.

foresterr, 2022-03-18:
Current implementation is less than ideal, marking every single wire joint with a dot.

Two wire joints are practically _never_ marked in any schematic or program, because it's pointless - there is no need to represent 2 unconnected dangling wires in any sensible schematic, and it's not even possible to draw something like that in Proto. Three wire joints can be either marked, or not (IMO not marking them is better as they are unambiguous, either its connected or it's a pointless dangling wire, same case as with 2 wires really, and also not possible to draw in Proto). Four wire joints are always marked, and I never saw a schematic with a 5 or more wire joint, but those probably should be marked too.

If this is followed, arcs are actually optional (still nice though), because it's clear that no dot means a crossing, not a joint. Unfortunately current implementation is still ripe for ambiguities, because it's quite easy to have 2 wire joint with another wire crossing over it, which looks exactly the same as a 4 wire joint.

foresterr, 2022-03-23:
Thanks, I get that - I was actually thinking about suggesting something along the lines of highlighting all possible connection points only after you tap some wire or component, but if you plan to fix it with a big change like that, I can wait. Hope your plans for autorouter include the possibility of connecting a wire to another wire at any point (only respecting the grid), that would be very convenient 🙂