Proto-App/Proto-Android #22: [Feature request] plugins

Ashwin-droid, 2021-03-03:
I think you should add a feature in which you could add components by third party plugins like if you want phase shifting transformer or a center tapped transformer you could install a plugin (like a pack of all types of transformer which you design or a third party) and expose some api to do that (which should be designable by a individual).

Ashwin-droid, 2021-03-03:
Or probably installing components pack from a url

Ashwin-droid, 2021-03-04:
@Proto-App please respond for #22 and #18

Proto-App, 2021-03-04:
For now we won't implement 3rd party components, simply because we don't have enough resources to develop such thing like thoughtful API. Furthermore our app is not popular enough and we prefer to develop more basic things during alpha stage.

closed 2021-04-22 by Proto-App

Ashwin-droid, 2021-04-23:
@Proto-App did you ment to close issue #4