Proto-App/Proto-Android #24: Unable to watch the transients on the scope properly

suryadeep56, 2021-03-07:
Hi, I found that you cannot go back to see the transients on the scope. I tried to simulate a simple RC circuit on it with reduced simulation speed. Is it possible to add something like where you can simulate it for a particular amount of time or go back the scope and have a look at the transients after pausing the simulation. 
Also, once the RC circuit reached steady-state, you cannot reset the circuit and it remains in the steady state. Please add a few features to observe transient circuits.

Proto-App, 2021-03-09:
Hello @suryadeep56 

Thanks for suggestions! We know that now you can not see historical values in scope, it will be available in addon (ability to record samples).

In second case we will add RESET button.

Proto-App, 2021-04-05:
RESET button is available in v0.23.0

closed 2021-04-05 by Proto-App