Proto-App/Proto-Android #27: Component request

waseemkhatri, 2021-03-14:
You may add new components like
1) IGBT
2) Triac
3) speaker/buzzer
4) DPDT switch
5) voltage regulators 78xx series or a universal voltage regulator that's value can be set by the designer
6) Motor
7) Diac
8) rectifier 
Etc...

64-Tesseract, 2021-03-16:
It'd be cool if the buzzer actually output sound through the phone speakers... but then you'd need to run your circuits in real time, which could be resource intensive...

Proto-App, 2021-03-16:
It should contain at least visual effect. But probably
some predefined sound will be availabe too (as buzzer setting)

MigueLobito, 2021-07-20:
> You may add new components like
> 
> 1. IGBT
> 2. Triac
> 3. speaker/buzzer
> 4. DPDT switch
> 5. voltage regulators 78xx series or a universal voltage regulator that's value can be set by the designer
> 6. Motor
> 7. Diac
> 8. rectifier
>    Etc...

Electrolitic capacitors, if they do work with polarity please let the symbol change to the respective one. I also support the idea of adding voltage regulators such as lm317 in order to simulate a standard 110v - 12v power converter.