Proto-App/Proto-Android #28: Component overlap bug

64-Tesseract, 2021-03-16:
I discovered that you can move and keep switches into an invalid position, and can move colliding components with their red highlight until you move a different component.
The circuit is completed, and you can even phase a switch right through another component if you move it dot-by-dot, but I'm not sure what other components this is possible with besides switches. I've uploaded videos demonstrating it where you can see my taps.

https://user-images.githubusercontent.com/63184095/111308966-71b4b880-86af-11eb-8513-9b7dafc71a18.mp4

Proto-App, 2021-03-16:
Thank you for report, we will try to fix it in next few releases.

Proto-App, 2021-03-22:
Fixed in v0.22.0

closed 2021-03-24 by Proto-App