Proto-App/Proto-Android #32: Capacitor in parallel doesn't seem to work

Ashwin-droid, 2021-04-06:
Please fix asap[image]

Proto-App, 2021-04-06:
Circuit is fine, you can not use ideal capacitors without resistance in simulation because it causes shorted loop with non zero voltage. You can add very low resistance in series.

Ashwin-droid, 2021-04-07:
Something still dosent work[image]

Proto-App, 2021-04-07:
Strange, our internal test shows that it works just fine:
[image]

Ashwin-droid, 2021-04-10:
Maybe it is due to capistance values

Ashwin-droid, 2021-04-14:
@Proto-App please try with my values

Proto-App, 2021-04-14:
@Ashwin-droid 
Yes, with big capacitor values it is not correct. We will check it.