Proto-App/Proto-Android #34: proto algorithms need to be fixed

bellmicheal663, 2021-04-14:
proto algorithms need to be fixed before adding new features or components.  Generic emiter amplifier circuit malfunctions, though bias like a textbook.  The mosfet power follower amplifier (Andrea Ciuffoli etc) circuits are not working and crash all the time.
Please add mirror components feature for bjt and fets.
[image]
[image]

Proto-App, 2021-04-14:
Thank you for report

bellmicheal663, 2021-04-19:
Can not give correctly predict the operation of any
[image]
 audio power amplifier circuit.

Proto-App, 2021-04-19:
@bellmicheal663 
Please provide reference waveforms and values related to your circuits, we have to compare them with our algorithm output.

bellmicheal663, 2021-04-19:
Sorry is not that picture.
[image]

Proto-App, 2021-05-07:
@bellmicheal663 
Could you provide correct waveforms and values for your circuit? We need to compare differences with reference data.

closed 2021-05-19 by Proto-App