Proto-App/Proto-Android #36: Source problem

svencanotgay, 2021-05-09:
I found a problem where when I use an AC source, voltage and current go only from one output. So my twoway replifier doesnt work right.

Proto-App, 2021-05-09:
@svencanotgay Could you describe your problem in more details? Or maybe post screenshot with your issue?

svencanotgay, 2021-05-10:
Sorry my description wasnt good. So I made a replifier. I used Gretzkys two way replifier. Then I connected filtration (capacitator) and zeners diod as a stabilizator. Then I connected a LED. This was ok and function was fine but then I blown LED bc there was a big el voltage. So I took down voltage and repaird LED. After this the whole device stopped working. Everything was fine but the current stopped flowing from the source through the one way of replifier. Everything was connected and components were fine too. So I made a same device under the old one and it didnt work eaither. So I had to made another one but in new sketch and this one finally started wotk. Sorry for my en I am not a native speaker ;)

Proto-App, 2021-05-12:
@svencanotgay Please post screenshot or short video.

Proto-App, 2021-05-19:
Missing details about this issue.

closed 2021-05-19 by Proto-App