Proto-App/Proto-Android #37: Copied component attaches to existing

KarlW100, 2021-05-09:
When a component is copied, depending on where the displayed circuit might be, it will be placed on top of an existing point attaching itself to the circuit. It cannot be moved without dragging the existing wires. The program should check before placing it it's going on existing points.

Proto-App, 2021-05-09:
Thank you for report, we will try to check if another terminal is present on grid.

Proto-App, 2021-05-12:
At first attempt we will try to check for existing terminals on circuit grid. Solution like dragging components from the panel sounds great! But from technical point of view it is hard because left panel is a native Android view component and circuit grid is very fast OpenGL surface and this feature requires some connection between two different views.

closed 2021-07-23 by KarlW100