Proto-App/Proto-Android #38: Component moves when error is generated

KarlW100, 2021-05-16:
If an error is generated while pressing and holding down on a push switch, the display will move up while the switch stays in place.
Sample video.

https://user-images.githubusercontent.com/75820960/118383947-18daa100-b5c8-11eb-9139-9f9293f28aac.mp4

64-Tesseract, 2021-05-16:
This might have been introduced in the fix for #29, where I had an issue with the error notification covering the add, pause, & delete buttons. Now the buttons move up with the notification, but so does the entire workspace.

Proto-App, 2021-05-16:
Looks like awkward behavior of touch surface. Not a serious bug but definitely case to improve. Thank you for report.

Proto-App, 2021-06-04:
Fixed in 0.27.0

KarlW100, 2021-06-04:
Perfect.

closed 2021-06-04 by KarlW100