Proto-App/Proto-Android #43: Delete simulation confirmation (Enhancement)

KarlW100, 2021-06-05:
Enhancement request. When the directory is displayed to load, copy, delete a simulation and the trash icon is pressed, display yes/no confirmation. Currently the simulation is instantly deleted. If the icon is pressed by accident (wanted to press copy but missed), all is gone.  The trash icon used to delete components and wires does not need the confirmation.

Proto-App, 2021-06-07:
Thank you for suggestion, we will introduce dialog with yes/no buttons or native Android UI component called Snackbar with undo button like below:
[image]

Proto-App, 2021-06-27:
Implemented in v0.28.0 (confirmation dialog)

closed 2021-07-22 by Proto-App