Proto-App/Proto-Android #46: It hurts

d-kanti, 2021-06-24:
https://user-images.githubusercontent.com/64345702/123229983-6f56cb80-d4f4-11eb-9b07-e63eaedb90ba.mp4

This hurts a little bit,  it should not go on forever... Not un real world atleast.  I think each wire should also have some resistance.... And also i think the inductor should show a voltage peak, whenever power supply is connected or disconnected.. the peak is absent....

d-kanti, 2021-06-24:
Ausom work by the way @Proto-App ....  Thank you for such a great app, have been looking for something like this for a long time.... Thank you so much.

Proto-App, 2021-06-24:
@d-kanti 
We will introduce optional resistance for wires (length dependent resistance)

d-kanti, 2021-06-27:
Again, Thank you, for such a beautiful app @Proto-App

closed 2022-09-17 by d-kanti