Proto-App/Proto-Android #48: Diode and LED has the same symbol

HyVong007, 2021-06-25:
I want Diode and Led has DIFFERENT symbol, The Led must have its own symbol.
[image]

HyVong007, 2021-06-25:
LED and Bulb: i feel hard to see if a Led/Bulp is lighting or not ?
How light is it ? May be some animations or some colors will help us easy to see the lighting.

Proto-App, 2021-06-25:
Yes, it can be confusing. We will add arrows for LED component. 
In case of bulb you have to set enough voltage and wait a moment, bulb should glow:
[image]

HyVong007, 2021-06-25:
Ok thank you. Im case the Led is burned by high current, Led should has another image or animation of Fire. It is the same with Bulp.
And the light of Led should be clearly to see all over the whole circuit. When i look at the whole circuit at some zoom level, i will see immediately the Lighting LED.
 Thank you

Proto-App, 2021-07-22:
Implemented in version 0.29.0 (LED has different symbol)

closed 2021-07-25 by Proto-App