Proto-App/Proto-Android #49: Dragging the screen causes app crash

HyVong007, 2021-06-25:
[Reproduce video](https://drive.google.com/file/d/1-1KyY6auTaD8BpUJ-7p93fm5Xe6JTusn/view?usp=drivesdk) 

Watch the video, at the end of the video, PROTO will crash.
Another issues here: when i am working in a circuit, when i want to touch a button/switch to open/close the button, i usually accidentally DRAG the button to another position. I feel annoyed why it is so easy to accidentally drag a circuit/drag a component ? I only want to click/tap/touch a circuit but always result in dragging the circuit.

Sorry for my bad english, i have tried to explain.

HyVong007, 2021-06-25:
Additional informations: my phone is Samsung Galaxy S8+ Exynos, Android 10. I use latest PROTO now.

Proto-App, 2021-06-25:
Thank you for report, we will check it.