Proto-App/Proto-Android #50: Improve graphic UI: show current value and voltage value at some special points

HyVong007, 2021-06-25:
[Watch Video here !](https://drive.google.com/file/d/1-bMLK64v2aa7T5LzwhKn3nSCQ2IoBqM7/view?usp=drivesdk) 

I want PROTO show more special important values in some special points, depending in some special components. For example: show voltage, current, direction, waveform icon (AC) at some point in circuit.

In case the circuit is CLOSED or OPENED, it will show some values at some points...

Why do I compare EVERYCIRCUIT and PROTO ???
=> I want to help PROTO better and better ! I love PROTO so much because i am a very poor student and Proto is FREE !

HyVong007, 2021-06-25:
In EVERYCIRCUIT, if i shake my phone, the circuit will receive some NOISES (EVERYCIRCUIT show Toast notification "INJECTING NOISES"). I think this is a great feature !
Can you add this feature to PROTO ?

HyVong007, 2021-06-25:
Each component such as: button/switch, LED, resistor, transistor.... Will have their special points. And when user creates a circuit with a component and start simulating, voltage and/or current values will be showed at their points.

Proto-App, 2021-06-26:
@HyVong007 Hello, thank you for suggestions, but we don't want to copy our competition. 

Our main goal is to provide free base product that will cover most of the cases with outstanding quality. You can expect paid add-ons in future because we need some financing to develop more advanced components and features.

HyVong007, 2021-06-27:
Ok sorry for my greeding. I will close this issues now

HyVong007, 2021-06-27:
I am too greedy. Sorry

closed 2021-06-27 by HyVong007