Proto-App/Proto-Android #52: Unable to solve joule thief circuit

corbym, 2021-06-30:
I am trying to emulate this circuit:

https://www.instructables.com/Joule-Thief-no-IC-and-no-Transformer/

I have set correct values for each component and also tried to compensate for the "perfect" components with extra resistors (not shown) but the circuit "cannot be solved":

[image]

corbym, 2021-06-30:
[image]

Proto-App, 2021-06-30:
@corbym 
Thank you for report, could you check with slightly lower time step (600ns) and increased LED current limit (100mA)? We don't have reference signal but we assume that LED should blink very fast.

[image]

[image]

corbym, 2021-06-30:
Yes this works, thanks 👍

closed 2021-07-01 by Proto-App