Proto-App/Proto-Android #58: A source request

Oky12, 2021-07-27:
Hi. I have a project that runs on battery and I want to try doing a simulation on this application. But I can't find the battery source. I'm trying to use a capacitor as a battery. But the capacity is too low to be used as a battery

I hope you can add that source.
And yes add some warnings like overcharge and overdischarge on battery source

Thank you for your hard work

TS232CZ, 2022-07-09:
I can see device battery but cannot set its voltage, just name :-(

Proto-App, 2022-07-10:
@TS232CZ 
It has 4.4 because it is voltage of your battery in phone. We have battery on our TODO list.

NikoFan01, 2024-03-11:
Hi, I wanted to know where you are with the battery and give you some tips. 
1) Adjustable capacity from 100mAh to 10000Ah; 
2) Adjustable voltage from 1V to 120V;
3) Maximum current (rated and peak) adjustable from 100mA to 10kA;
4) Default self-discharge current depending on battery type (lead-acid, LiFePO4, Lithium-Ion, ...) also based on the ambient temperature, which can be adjusted manually, and that if charged too fast or with an already high ambient temperature, it slows down the charge itself by increasing the internal resistance, which can also be adjusted according to the type of battery.

Thank you and good luck,
Niko

Proto-App, 2024-03-20:
Hello @NikoFan01 
It will be probably model with basic characteristics without ambient temperature etc. because we don't have any battery characteristics in different environments.

NikoFan01, 2024-04-04:
Hello, the temperature of the battery in the case of lead-acid and lithium batteries affects the maximum power delivered or drawn. For example, if a lithium battery overheats the maximum current is limited by the increase in internal resistance, And if the temperature rises even more or stays high for a long time, the battery ceases to function because they degrade internally with the possibility of ignition.

NikoFan01, 2024-12-17:
Good evening, I would like to know if you have made any progress regarding the battery to be implemented in the app, thanks in advance for any answer. 
Niko

Proto-App, 2024-12-19:
Hello @NikoFan01 
Not yet, we have order of next components to implement. Currently we are working on named terminal component. By the way you can try to implement battery in JavaScript (simulate internal behavior).