Proto-App/Proto-Android #59: [bug] Artistic Circuit Preview

VincentWolf, 2021-08-13:
i've some artistic preview Instead of reality :  
[image]

Proto-App, 2021-08-14:
@VincentWolf 
Please write steps to reproduce this issue. You can give more details like Android version, phone vendor etc.

VincentWolf, 2021-08-15:
``Device Info v2.5
©ToraLabs
Created on Sun Aug 15 10:54:31 GMT+02:00 2021

DEVICE
-----------------------------------
Model : View2 Plus
Manufacturer : WIKO
Brand : WIKO
Device : W-P210
Board : p210cn
Hardware : qcom
Android Device ID : – Do you Really need it? –
Device Type : GSM
WiFi MAC Address : 44:D3:00:00:00:00
Build Fingerprint : WIKO/W-P210-OPE/W-P210:8.1.0/OPM1.171019.019/androi12:user/release-keys
USB Host : Supported
Google Advertising ID : 7fdaf832-49b6-4dc8-XXX-e18646502dXX
Timezone : Central European Standard Time (GMT+02:00)

SYSTEM
-----------------------------------
Android Version : Android 8.1.0 Oreo
Released Date : December 5, 2017
Code Name : Android 8.1.0 Oreo
Security Patch Level : 2018-12-01
API Level : 27
Bootloader : unknown
Build Number : OPM1.171019.019 release-keys
Baseband : .TA.2.3.c1-00705-8953_GEN_PACK-1
Java VM : 2.1.0
Kernel : 3.18.71-perf
Language : English
OpenGL ES : 3.2
Root Access : No
Treble : Supported
Seamless Updates : Not Supported
SELinux : Unable to determine
System Uptime : 210:49:42
Google Play Services : 21.26.21

CPU
-----------------------------------
Processor Name : Unknown
CPU Hardware : Qualcomm Technologies, Inc SDM450
Supported ABIs : arm64-v8a, armeabi-v7a, armeabi
CPU Architecture : 8 x 1.80 GHz
Cores : 8
CPU Governor : interactive
CPU Type : 64 Bit (aarch64)
CPU Scaling Driver : msm
Frequency : 652 Mhz - 1804 Mhz
Running CPUs : 
   Core 0       1804.80 MHz

   Core 1       1804.80 MHz

   Core 2       1804.80 MHz

   Core 3       1804.80 MHz

   Core 4       1804.80 MHz

   Core 5       1804.80 MHz

   Core 6       1689.60 MHz

   Core 7       1689.60 MHz
CPU Usage : 100%
BogoMIPS : 38.40
[... 813 lines cut ...]
com.tinno.hardware.display.camera_notch
android.hardware.location.gps
android.software.midi
android.hardware.wifi
android.hardware.location
android.hardware.vulkan.level
android.hardware.telephony

DRM INFO
-----------------------------------
• Widevine CDM     
Vendor : Google
Version : v5.1.0-android
Algorithms : AES/CBC/NoPadding,HmacSHA256
System ID : 44455
Security Level : L3
Max HDCP Level : 
Max No. of Sessions : 16
Usage Reporting Support : True
HDCP Level : 

• Clearkey CDM     
Vendor : Google
Version : 1.0
``

VincentWolf, 2021-08-15:
> @VincentWolf
> Please write steps to reproduce this issue. You can give more details like Android version, phone vendor etc.

I just install this app from Play store and run it... Nothing else... 
My phone hav't any sd cards but he has some linked directories to emulate extern disk space. It's the only idea I had at this time (i send new info if i found it...)

VincentWolf, 2021-08-16:
Another info :  All pictures in example tab works well.
Visually and after thinking... it's look like a simple problem : 
It's like the width of pictures zones are smaller than preview and when they are drawed, each pictures lines are breaks before the real width and all undisplayed pixels are puts at the start of the next line like "word-warp" in a text app.
It's really displayed like that, after some few seconds of contemplating, I've understand how this app can display this jpeg and after we see this "scrambling", the apk <answer to me> “why you don't understand all my previews pictures? They are the same like all other sprites„..
😅🤣

Proto-App, 2021-09-03:
We can't reproduce this issue on our devices. Maybe something is wrong with Android driver on Wiko devices, it has pretty old SOC (Qualcomm SDM450)

Proto-App, 2021-12-30:
Can not reproduce on Nexus 4, Realme Pad, Xiaomi 9T, Xiaomi A2 Lite, Samsung Galaxy TAB

closed 2021-12-30 by Proto-App