Proto-App/Proto-Android #60: [ideas] Somes ideas...

VincentWolf, 2021-08-13:
Can you add a components database which contains values and firmules and the link to the circuit (ex: 2nXXXX, LM78XX)
For help to create some components create some elements like "Heat" with the thermal resistance and working temperature. And another element "Burn" that we can set conditions of unknown results of this component.

ps:
Excuse me for my lack of English...

Proto-App, 2021-08-14:
@VincentWolf 
Sorry but we don't understand your request.

VincentWolf, 2021-08-16:
I try to translate all my «Personal Fr-English » writed previously... and after all existing bugs, ideas... hem... after the next version of your apps, i will send you <again> my ideas with my dictionary...he no.... with the real English text

VincentWolf, 2021-08-16:
If i found some unused computer time and one useful hand (and attached to my own body, it's more simple when we use this )

closed 2021-08-16 by Proto-App