Proto-App/Proto-Android #61: Wire joints

gdosh, 2021-08-14:
I don't know if this counts as an issue, but say I have a wire running between a capacitor and resistor.
Now when I try to attach another wire to the the midpoint of this wire (perpendicular) it doesn't work. 
I have to delete the wire, recreate the wire with joints (dots) midway, so that the new wire can now connect to the older one. (Perpendicular joint). 
Im guessing this is to avoid a short (cross connection) when wires overlap. 
Also a 'BACKSPACE' / UNDO button would be nice.😅😅
Anyways thanks.
This is a brilliant app.

VincentWolf, 2021-08-15:
I've the same problem... i'm tired to try some hints to create my program...
I'm in good way to finish my project... Do You think so'?! 
[image]

gdosh, 2021-08-15:
Wow. What u created looks awesome. 
But I'm not an electronics expert. I can't really decipher circuit boards. I just tried proto because I wanted to test out a power supply circuit I'm trying to fix.

VincentWolf, 2021-08-16:
No, it's just an "Folower aop" (litteraly the translation grom French)... Linear OPA With a RCLL pass band (active) filter. i create a list of some frequency generator from basic sinus to radio fm modulation... And... I don't know why... I've created some good looking utils wire... but many other  useless wires that come outside like it from my complex cortex and this two brains...