Proto-App/Proto-Android #64: Capacitors with μF/mF Decimal values

YukarinnAM, 2021-08-20:
Hi, i don't know how should i report this
Because i think is an interesting error or an unused function maybe
But i saw that when putting manually by input the values of Capacitors on Farads you can induse a decimal point on mF μF and all the others that you can't manually put with decimal point (example you can't put 2.2mF)

The error is easy to induse you just put the value on F(arads)
And write a number (12 example)
Then go back to the start of the input without deleting the number and proceding to add 0.00 , this can make the 12 move but not deleting it's value
[image]

I'm sorry if it's hard to comprehend 
English is not my mother language

Okey once you have a smaller value than the decimal limit (0.001) you apply that value and tada, you have 1.2mF

[image]

I think that is not really an error because sometime i want to have capacitor with value like that without having to put another in series with it (like 2.2mF for Full Bridge Rectifiers) 
This i belive can be easily solved by just allowing the user to input decimal values on μF and mF, because i saw some capacitors in real life with 1000μF and 2.2mF

Thank you very much for putting your effort on developing this app, it goes for everyone who is involved in it

YukarinnAM, 2021-08-20:
Instant update of the problem
I proved by testing that it indeed works
A capacitor with 10.1μF virtually max charge last a little longer than a capacitor 10μF virtually max charge
Here

[image]

As you can see, at first it wasn't a lot of difference 
But then it was an evident difference

YukarinnAM, 2021-08-28:
No prob, also i found out that you can do something similar with inductors and resistors

HubbsInHobby, 2021-09-19:
I think, decimals are so important (at whatever prefix of standard unit used). I'm using home-made HV-HF capacitors that always have a decimal... What I do to get around this problem is to use the copy-paste function. But I hope decimals can be done easily with any electronic components.

Yet, I like the "feel" of this simulator! Keep it up!

Proto-App, 2023-04-08:
@YukarinnAM  @HubbsInHobby 
Implementd in version 1.18.0