Proto-App/Proto-Android #66: Insufficient workspace dimensions

Raart, 2021-09-03:
After playing around with Proto, I experienced that if I will come to more complicated circuits there is a serious workspace problem.
Symbol dimensions are too wide to fit onto the workspace.
Please add greater virtual space and/or pinch/zoom to change scale ratio to get the correct visual and workspace needed.

Proto-App, 2021-09-22:
@Raart 
Do you have any suggestion about workspace size? We can not make it unlimited. Bigger workspace can introduce performance bottleneck in some cases (slower phones)

Ashwin-droid, 2021-10-06:
Yes, you cant make unlimited workspace beforehand (nothing is unlimited!!) 
But there is a exception,
if you reach the end of your workspace, It will render another 3-5 blocks of space on that side so that on powerful phones The space is virtually limitless 
And on old phones they have to limit it (Anyways the space was limited in the first place)

FREE Optimizing trick :  Remove the unused blocks ( like if the circuit is going on the side, you can unrender (free up memory) top and bottom blocks ) this method will save memory exponentially as circuit complexes

closed 2022-04-01 by Proto-App