Proto-App/Proto-Android #67: Weird Increased Voltage with fuse

YukarinnAM, 2021-09-07:
Hello, i have found a weird error that occurs when using a fuse and some lamps
The fuse brokes because of a weird increase on the current 
[image]

Here we have ththe circuit working
To replicate do exactly this
Add the ac voltage source(don't change the voltage yet), connect a 10A 100W Fuse to the AC Voltage Source 
Then add a switch and put some lamps (like 5 or 6) in pararel
Connect it to the ground of the voltage source and activate the switch
Then go slowly increasing the voltage of the AC Voltage Source up to 240 or 220
You should see like there is no problem 
Like 3 to 4 amps is running throught the fuse and it doesn't turn
And that's okey, that's normal
Well
No try pressing the reset button or to turn the switch off and wait until the lamps cool down then turn it back on
[image]
[image]
[image]

Exactly, the fuse blows up
And if you try to lower the speed and the time step after the lamps cool down
You can see that if you turn the switch back on, a weird huge current runs throught the fuse for some weird reason
Is like it has a Capacitance or something i'm not sure

[image]
[image]

I wish this report helps to improve the app even more
Once again
I thank the proto team for putting their effort into this great app
Btw, i'm looking forward for the subcircuit system i readed you are developing

Proto-App, 2021-09-09:
Thank you for report, looks like bug because current is way to high.

YukarinnAM, 2021-09-23:
Ohhh i see, that sound logic, i saw that if you turn off at switch and let it cool just for it to stop emitting light and then turning it on,it doesn't blow the fuse
Also that Power setting sound great to get more realism on the bulbs

Proto-App, 2021-10-08:
Power setting implemented in version 0.33.0