Proto-App/Proto-Android #68: Visual Transformer Voltage Error

YukarinnAM, 2021-09-22:
Hello
And sorry for once again bother you devs with another bug
Now i found a inoffensive bug 
I discovered it like this
I was making a Full Bridge Wave Rectifier using a transformer
I wanted to see the Amperage of the AC Source so i did that
(Image is when i tried to replicate the bug, it is indeed replicable)
[image]

Using the graphic option of the amperage
Everything is right, no problem at all with that
The problems happened when i tried to see the voltage i got on the transformer secondary
[image]

I saw that you can actually automatically change the graphic to measure Amperage to Voltage of all the marked components
So i thought this will work but then i saw something weird
[image]

The Value was correct, but not the symbol(not V but A?)
When i turned the AC source mark off, the Symbol became voltage
My theory is that this happens because the amperage and the voltage graphics are draw one up the other
[image]

Sorry for my bad english
My biggest wishes to the app
Is going to in just the right way to become something students can use to learn with pseudoPractical Exercises
And please when the moment to make some functions paid, make the price affortable enought for you guys to receive what you deserve for such a great app, but also for people or students of around the planet to be affortable as a learning material or even as a autodidactic learn material

Proto-App, 2021-09-22:
@YukarinnAM 

Thank you for report. We will fix it in next release. 
In case of addons we want to make features affordable for everyone :)

YukarinnAM, 2021-09-23:
Thanks, remenber to take care to keep upgrading those features efficiently