Proto-App/Proto-Android #69: Component suggestions/requests

t4st3r, 2021-10-07:
Hi, 
I really enjoy this app. There are a few missing things for me to use it properly: tapped transformer, custom transistor values (not that crucial) and also i encountered this weird MOSFET issue where it's not fully open until about 100V of GS voltage, which is unusual at least. 
Otherwise, thank you for your hard work, this is AFAIK the most accurate and realistic simulator for Android i have ever used.
Have a nice day!

closed 2021-10-29 by Proto-App

caiohamamura, 2021-12-14:
Same here, why the MOSFETs don't have a settings button to set the Vth?

t4st3r, 2021-12-14:
> Same here, why the MOSFETs don't have a settings button to set the Vth?

yeah that would be really nice