Proto-App/Proto-Android #81: Capacitor unable to be set at 100μF

YukarinnAM, 2021-11-09:
I don't know why or how this happens but for some reason
The user is unable to set a Capacitor to 100μF
If you set it manually it automaticly sets to 99μF
[image]
This only happens with Writed User Input 
Is really easy to test just create a new simulation, add a capacitor then try to set it to 100μF
It should appear as 100μF but when using the user input setter it shows 99μF
It isn't a bug of amount of chars of that value either
Since you can set it to 101μF

Proto-App, 2021-11-10:
Thank you for report. Looks like some silly bug... We will check it.

YukarinnAM, 2021-11-10:
Thanks for your constant support!
Also
I found out that the error is actually a single visual bug
Since i tested the charge/discarge voltage being both idealy at the same time and the result is that there was no discharge difference for a whole μF of difference
Also
Thanks for the whole new content that comes on addons
I'll save some money to support this great app!

Proto-App, 2021-11-11:
Thank you! Please share app with friends or rate app on Google Play, it can help a lot :+1:  You can expect more new content with incoming releases (also free components)

Proto-App, 2021-11-25:
Fixed in version 1.1.0

closed 2021-11-29 by Proto-App