Proto-App/Proto-Android #87: Fuse not blowing when Power is reached

YukarinnAM, 2021-11-14:
Hi! I found that if you try to make a circuit with a fuse and even if you pass the Power limit of the fuse on a component of the circuit, it still doesn't blow as expected
Is not that the Power limit doesn't work
Is like it doesn't detect the power peaks on the circuit
If we put a Fuse with 10kA limit and 1mW Power it still blows if you pass a voltage high enought to not reach 10kA but surpase the 1mW power limit
But i think this error happens because of the internal fuse resistence, it acts as a voltage divisor and makes it not detect properly when the circuits power is more than the power limit
[image]
[image]
In the images you see that the Led power is more than 30 times the fuse power limit
Also thanks for the free digital one! ^^
I saw that the app got a small boom when the addon update was released
This is taking the right way

YukarinnAM, 2021-11-14:
[image]
[image]
My theory was correct
The small resistence of the fuse acts as a Voltage divisor with the rest of the circuit

Proto-App, 2021-11-14:
@YukarinnAM 
Thank you for report. Interesting case, we have to figure out how to cover this case.

YukarinnAM, 2021-11-14:
I'd been thinking about this error for some minutes and questioned
Is power constant on a circuit?
Tecnically over the whole circuit yes
It is
But is it over a single resistor or part of wire
When we have voltage divisors, the voltage is reduced depending the resistence diference,wich means. That the voltage is not constant all points, or between some point, wich mean, V * I = P , and only current is constant, P is not a constant on a whole circuit, irl, that means that fuses can't be build to work on a specific Power limit but only on a Current limit, maybe the problem is not the fuse but the concept of a Power setting
More research needed to confirm that the power configuration is not really usefull or actually able to be created irl

YukarinnAM, 2021-11-14:
What i mean is that
You could have 10kV over a resistence but it won't burn out if the current doesn't surpase a limit in wich the joule effect starts to break it appart
So technically the power setting is useless and impossible to fisically produce ,the power limit on resistence is the limit of when the joule effect starts to break the resistor
At least on term of how the fuse works, a power limit is just sadly in vain

NMrocks, 2022-02-04:
> only current is constant, P is not a constant on a whole circuit, irl, that means that fuses can't be build to work on a specific Power limit but only on a Current limit

This is it. Fuses are not supposed to work on a specific power limit but instead on a specific current limit, because a fuse's job is to avoid excessive current in a circuit. The option for setting the max power for a fuse should be removed from the app IMO

Proto-App, 2022-02-07:
After your explanations and some internal discussion we decided to remove power setting, as you mentioned it does not make sens in this case. You can expect it in next release.