Proto-App/Proto-Android #89: Function request, Invert Face

YukarinnAM, 2021-11-18:
Hello!
I want to request a very usefull function 
Especially when working with transistor
The function is to invert the view of the object
Like the C E pins on transistor
Also the side of a relay and like that
This because i find very hard to make complexier circuits with transistors since i have you have to put too many wire and the circuit get very messy as more transistors you put, just inverting the pins is okey
Not really needed the function for relays ,555s or Opamps for now
Wish this can be impremented
Again, excuse my bad english
Thanks

YukarinnAM, 2021-11-19:
Also an idea for the polarized capacitor
You could give it a resistence setting, if the resistence is 0 then the capacitor is on "ideal" mode

Proto-App, 2021-11-19:
Yes, transistors will get mirroring in near future.

Proto-App, 2022-01-16:
You can flip transistors in both axis in version 1.4.0 :)

closed 2022-03-23 by Proto-App