Proto-App/Proto-Android #90: Add ability to rotate integrated circuits

Lkn7, 2021-11-19:
I bought the packs and noticed that i couldn't rotate the chips like the other components.
Please add that ability. Otherwise great app!

Proto-App, 2021-11-19:
@Lkn7 
Thank you for support! We would like to implement flipping (mosfets, transistors) before rotation ability in all components.

JunkMcNuggets, 2021-12-17:
So do you have to buy the add-on packages just to be able to flip the components over in the other direction, Or is It just not implemented at all yet? Is that why the little double arrow icon on the bottom left doesn't do anything except spin the little arrows when I hit it?

Proto-App, 2021-12-17:
@JunkMcNuggets  This button can stop and run simulation and it is works as expected. We have already IC rotation in our internal test track, first component with this ability will be RGB Led.