Proto-App/Proto-Android #96: Capacitance units

Agent259, 2021-11-28:
Can you add adjustable capacitor units above 999
Right now its only available upto 999 to manual entering
And
Remove mF from the capacitor unit selection it's refer to micro farad μF 
Both are same
[image]
Pico farad pF , nano farad nF, micro farad μF and farad F
Most of the brands use μF as mf

Agent259, 2021-11-28:
[image]

Proto-App, 2021-12-02:
Hello @Agent259 
m - mili -> 0.001
u - micro -> 0.000001

Yes, we have wrong symbol (u -> µ) we will try to improve it.

Proto-App, 2021-12-09:
Fixed in v1.2.0

closed 2021-12-14 by Proto-App