Proto-App/Proto-Android #97: Text duplicate doesn't preserve color and Circuit recomendation

YukarinnAM, 2021-12-01:
Hello, i was making a recommendation for the circuits you have as samples, a small but easy to comprehend Negative Gates (NOT , NOR)
[image]
[image]

Is a bit interactive by turning the switches and you also have a truth table under both gates 
Well,while i was doing this, i also saw that if you duplicate a Colored Text, It loses it color and goes back to white
Seems like the same error as the previously existent "Power setting on the bulb bug"
To replicate it , just add a Text, change its color and duplicate it

Beside that small bug
I hope you guys found interesting enought this simple circuit to be added on the sample circuits
Thanks for you time once again

Proto-App, 2021-12-02:
Hello @YukarinnAM 
Thank you for report, we have almost identical issue with spark gap :rofl: sorry for silly bug, both will be fixed in next release.

By the way great circuit! We will introduce share function probably in release v1.3.0, could you share your circuit with us then?

YukarinnAM, 2021-12-04:
good luck with the bug also of course
I'll honored to help with this example
Also hyped up for the share circuit function
But i somehow feel like that can be easily exploiteable to use paidComponents for free with the duplicate function beware with that

Proto-App, 2021-12-05:
@YukarinnAM 
Don't worry, this case is secured :) user will get message about missing component during simulation.

Proto-App, 2021-12-09:
Fixed in v1.2.0

closed 2021-12-14 by Proto-App