Proto-App/Proto-Android #102: 7 Segment Decoder no parameters

Gralloc, 2021-12-13:
The 7 Segment Decoder has no way of changing the output states, so A to F are always set at high 5v G and the other outputs are always set at 0v, this makes the decoder near useless.

Proto-App, 2021-12-13:
@Gralloc 
Thank you for report, we will verify it.

Proto-App, 2021-12-14:
@Gralloc 
Could you post screenshot of your circuit? We verified it on our side and it works ok, check screenshot below:
[image]

Gralloc, 2021-12-14:
Hi, guys, guess what, you are correct, it was my fault and I had made a bad connection and did not spot it. 
Thanks for your quick help, and for this software app, it will be a great help in getting me back into electronic circuits after a very long  illness. 
Cheers
Graham

Sent from the all-new AOL app for Android

closed 2021-12-14 by Proto-App