Proto-App/Proto-Android #103: Current and power is mismatch

Agent259, 2021-12-22:
Bulb is drawing more current then fixed value 
As i set 25W current for bulb but it's drawing more then as i fixed
[image]

Proto-App, 2021-12-22:
@Agent259 
Thank you for report, we will check it.

YukarinnAM, 2021-12-28:
Actually
I believe that the bulb just acts as a resistor
You set the voltage and the power
And it defines it's resistence and some inductance
That's why the power and voltage and current is sometimes missmatched
Imagine you have 25W bulb
And with the setting on 25V
That means 25W / 25V = 1A wich the simulation makes it be like  25V / 1A = 25Ω
So the bulb resistence is 25Ω
Nothing in the real world can consume an especific Wattage , is only a resistence tested over the common use voltage