Proto-App/Proto-Android #104: Circuit example for more gates, also new error

YukarinnAM, 2021-12-28:
Hello!
With the new update
And because you guys told me to give you this when this update came out
Here you have the more gates circuit example
https://www.protosimulator.com/share?circuit=More_Gates_1640735118843
[image]

Also, would like to show a new error, that's most likely because it creates an .zip of the circuit wich ends up getting detected as a "circuit"
[image]

For some obvious reasons i believe that i didn't created the .zip

Proto-App, 2021-12-29:
Hello @YukarinnAM 
Thank you for circuit! We will add it in next update.

In case of issue ;) Could you share any scenario to reproduce this case?

YukarinnAM, 2021-12-29:
Thank you very much for your work!

Of course
All i did was actually simply press the share button, haven't pressed again tho
Could maybe be a only once error
Also, once i entered again on the up
It seems like it dissapeared

Also, i tried to repeat the steps i made
Wich was simply press the share button
But doesn't seem to appears again, weird
Sadly it seems there is no way to reproduce the error, perhaps trying many times

Proto-App, 2021-12-29:
Ok, probably we have one scenario when it can be possible :+1: 

By the way, we would like to use your nickname and put it in description (author label), is it ok for you?

YukarinnAM, 2021-12-29:
Oh, of course
I have no problem with that ^^
I would be honored!

Proto-App, 2022-01-16:
Example is present in version 1.4.0.
We fixed issue with zip visibility in one case.

closed 2022-01-24 by Proto-App