Proto-App/Proto-Android #110: two fingers "pinch" gesture on scope for changing display timing.

Bushmills, 2022-01-16:
I'm currently using a combination of simulation step and simulation speed for getting the scope display right, not sure whether that's supposed to be done that way.  However, being able to allow by gesture to narrow or widen the displayed scope traces will make it clear that that'd be the proper method to do so.

Proto-App, 2022-01-17:
Yes, we have such feature for future update.

foresterr, 2022-04-21:
Pinch to zoom would be nice and very mobile-y, but a way to precisely set scope's time base by just inputting it is even more essential, imo.

Bushmills, 2022-04-21:
Zoom per gesture was of course implying (albeit not explicitely saying so) that time per division, as you're used from a real scope, would be shown too.

Proto-App, 2022-06-13:
Implemented in version 1.8.0

closed 2022-06-13 by Proto-App