Proto-App/Proto-Android #111: Dim light on led without any current flow

YukarinnAM, 2022-01-18:
Hello, a small bug i found
Is that leds have a permanent dim green glow
Idk if this is supposed to be like this or no
But i report this just in case
Still is not really a problem and could be usefull actually
[image]

Also, i'm so happy to see the flip function ln transistors
That really is a life savior when it comes to make the circuits more visually compact

Proto-App, 2022-01-18:
Thank you for report. It behaves as intended, we wanted to show some visual aspect (this is also present in led matrix, 7 segment led and led bar).
In case of transistors it greatly simplifies some circuits and it was easier to implement than we expected :) More components will get this behavior.

YukarinnAM, 2022-01-18:
Ohh i see i see
The led dim color actually feel like it is the filter plastic on a real one, so looks great

Also, yeah, fliping some other components would also be awesome, i think relay is the one that needed the most after transistor
Good to see this much progress from just around 4 months, keep the great job! ^^
Also: any idea of how or when will be the subcircuits be impremented?

closed 2022-01-24 by YukarinnAM