Proto-App/Proto-Android #116: A lot of Recommendation/Requests: NonIdeal components

YukarinnAM, 2022-02-05:
This is something that most people would apreciate 
since designing some circuit is harder when the components needs resistors or in case of the mosfet a capacitor(usually 1 to 10 nF) 
A good start would be a non ideal Comparator since all it need is a V+ and V- Pins
Capacitors are a problem as well, but with the oncoming wire resistence it should be okey for now, mosfet is only a problem in cases where you need it to stay on

(in case anyone read and need this situation, just add 1 to 10nF capacitor from gate to source, read your datasheet and put a capacitor in case you want to simulate high frecuency scenarios)

Making more ICs is also something that would be really appreciated
i also thought about a system where you could make a circuit, and turn it into an ic where you choose where to put the pins etc, and make it something that you could share with other users to have a decent amount of ICs and maybe even Custom ones
Althought this is in my opinion a Great idea for ICs iversity and attractive for circuit designers or to test different ICs like CD4047, TDAs , AudioAmplifiers, the app is still in on early age, and need many touches before something like that, also it will probably be a PAIN in uppercase for you guys to do the programming and inApp design of a whole new system like that and the problems it can lead, thought  i would love it to be free for everyone and so we could something like a website to share the Circuits turned already into ICs for us to just copy or download and enjoy making circuits with , but would also be great if you all get reward for it, so leaving it free or for paid would be your option
but i'll leave the concept there, hopefully in a future ,maybe a bit distant , it can be on the roadmap

also Ohmmeter
yeah, An ohmmeter
why? is kinda usefull when you need to know resistance of many resistor mixed in series and parallel something like this 
[image]
The math is actually just the basic Ohm Law (R = I/V)
Just send a small voltage throught the circuit like an irl ohmmeter does, maybe around 3V? or with a voltage setting, and it reads the Current and Voltage used, with that showing the overall resistence I/V

Again i want to say, all this is only recommendation and just giving concepts/Ideas that i personally found usefull but also think that would be usefull for more people

Audio input i saw on the roadmap
Making an Audio Output would be really cool by using integrated microphone,but please if that gets to be an idea limit the max volume it can deliver so we don't have a 10kV 10GA 100% volume kaboom on the phone,fixed 2 4 6 8 16 Ω maybe?

Anyways, remenber once again that is all my opinion and just throwing cool ideas, maybe the only ones for know possible are the ohmmeter and the ideal Comparator
Have a great day and thanks for the hard work like always!

Proto-App, 2022-02-06:
@YukarinnAM  As always thank you for feedback!

We have ohmmeter in internal test track :) and it should be available in next release. 

There will be a share option for custom ICs and circuits. User will get one additional screen with all circuits and ICs available to download :) But it will be available only for users with PROTO account (users with BUNDLE pack or with 3 unlocked addons, this feature is in progress). We want to limit acces only to user with personal account because as a owner we need to moderate uploaded circuits. Furthermore this feature could cause huge server load if it will be available for everyone so we have to limit possible risk. 

And about custom ICs, this is great feature (we called it subcircuits) and we want to make it available for everyone but it is in development (early stage).

Bushmills, 2022-02-06:
Often the problem with ideal devices when simulated is that they or the circuits they're used in stay in an equilibrium state, a balance needing some extra effort to offset it. This "kick" can often be achieved by simply - if I'm allowed to use this word - adding noise to resistors, as those are commonly used about everywhere. Their noise will then propagate, and affect other parts of the circuit, aiding in throwing them off balance.
So by upgrading (or should it be downgrading, after all the new state would be "less perfect") a single component type, many cases of too perfect devices could be covered already.

Bushmills, 2022-02-06:
I'm looking forward to subcircuits (or "blackboxes" or "complex custom components" or whatever they may be called) - I think those may combine the best of the two worlds "increasing circuit real estate" and "keeping circuit complexity manageable".

Bushmills, 2022-02-07:
I'd use a value which is a percentage of its resistance in case of resistors. But your idea of applying this to wires instead (or additionally) is very good. With those I'd also consider to introduce a very small resistance and let it slightly fluctuate, though it's effect on simulation performance may be considerable (the resistance of wires, not the fluctuation), with the result that I might keep perfect wires, grudgingly, and noisify resistors, as I need to take account of their resistance already anyway.
For the amount to fluctuate I'd try a value which may be in the region of about one percent of the tolerances these resistors could have in the real world, thus around 1 percent of 1...5 percent. Important may be that there is some fluctuation at all, not the actual extent of it: any disturbed unstable equilibrium has the tendency towards progressing away from equilibrium state - the actual value then determines mostly how quickly this will pick up, and this again depends on simulation speed.
A configuration setting of resistor tolerance may be a good approach, as changing this could then also affect the amount of noise, allowing user to customize it according needs, and even to set up a single resistor to become a potent random source, if so desired.

Proto-App, 2022-02-12:
@Gertjr1999 New release will be available in next week and you can expect more components :) by the way please create new issue related to 4017 ICs because this thread is connected to other topic.

YukarinnAM, 2022-02-12:
Good to see this thread have some interesting ideas like the resistors noice

also something more that would be really apreciated would be a Light Mode
because when you are outside, the reflect on the screen and the sun makes the dots and the limit lines of the workspace practically invisible for the eye
maybe just inverting all the colors but Current,Oscilloscope probes,Bulb and led lights and voltage can work

Gertjr1999, 2022-02-24:
> @Gertjr1999 New release will be available in next week and you can expect more components :) by the way please create new issue related to 4017 ICs because this thread is connected to other topic.
> 
> EDIT: You can always check our roadmap: https://github.com/Proto-App/Proto-Android/wiki/Product-Roadmap

Soo its been 12 days , didn't have an update yet?

YukarinnAM, 2022-02-24:
Sometimes google doesn't update automaticly
but the new update is out already
I saw that we have a new component pack

Proto-App, 2022-02-24:
@Gertjr1999 
No, we have to work with translations, it is postponed. 

@YukarinnAM 
Do you mean new pack (Analog pack?) this release has been available for 3 weeks

YukarinnAM, 2022-03-05:
oh
well
this is what i mean about google not updating sometimes for me