Proto-App/Proto-Android #124: More examples with 555 ic

MontMontgomery, 2022-03-15:
[image]

[https://www.protosimulator.com/share?circuit=555_IC_1647378107652]()

3 basic operation mode
Astable, monoestable, FF SR

Good work on creating this app!

Proto-App, 2022-03-16:
@MontMontgomery 
Thank you for circuits! Can we use your nick name as an author name in description in PROTO?

MontMontgomery, 2022-03-16:
> @MontMontgomery 
> Thank you for circuits! Can we use your nick name as an author name in description in PROTO?

For sure!

MontMontgomery, 2022-03-16:
[https://www.protosimulator.com/share?circuit=555_IC_1647441806904]()

Some minor improvements

closed 2022-08-11 by Proto-App