Proto-App/Proto-Android #131: Audio output

MontMontgomery, 2022-04-06:
That's it, just another component.
I saw that Audio Input has been added, so i think Audio Output should be a great idea

foresterr, 2022-04-16:
Definitely. There was another issue for mic and speaker, but it was closed after mic and audio input were added.

Audio out to wav file would be great, live audio out to speaker would be awesome.

MontMontgomery, 2022-04-16:
Yeah, and a buzzer should be great

kjcnett, 2022-06-12:
Yeah definitely could use an 8ohm speaker for audio circuits.

electron271, 2022-06-28:
Would be very useful

TheRektafire, 2022-07-27:
Yeah but the DAC is locked inside the misc pack so it would be behind a paywall presumably 🙃 but I guess it would be nice to have nonetheless...

electron271, 2022-07-27:
We have an ADC, would it really hurt to give us a DAC?

TheRektafire, 2022-07-27:
> We have an ADC, would it really hurt to give us a DAC?

if you look on the list of features you'll see there's a DAC but it's part of the "analog pack" and costs 3.99 :( presumably the audio output would be put in there too sadly

electron271, 2022-07-27:
$4 may not be much, but I think a DAC might be a bit more useful than an ADC in some cases