Proto-App/Proto-Android #132: Phase of AC Voltage source returns odd value on 30°

YukarinnAM, 2022-04-08:
Hello!
I was doing some phase tests and i saw a weird return when setting ac source phase to exactly 30°, it returns 29.99999999999996°
other numbers seems to present a similar behaviour with a very small difference chance
I tried to replicate the bug
the steps are the next ones:
1) New proyect or existing
2) Add AC source
3) Select it
4) Change phase to any number (i tries 30,150 and, but by the behaviour of the bug should work with any but 0(maybe))
5) unselect it(important, touch any other place on the screen)
6) Select it and see the phase on the manual set
it should appear aomething like this
[image]

i found this by chance and is not really more than a estetic thingy or highly precise phase problem
but i suppose that reporting it is a good idea just for the sake of informing 

Thanks for a great app
also finally brought the bundle pack like a month ago, it really comes a lot of great stuff
thanks again for a great work!

Proto-App, 2022-04-10:
Thank you for report, weird behavior.... we will check it :+1:

Proto-App, 2022-05-05:
@YukarinnAM Fixed in v1.8.0

closed 2022-05-05 by Proto-App