Proto-App/Proto-Android #139: Frecuenciometer Request and Power on Osciloscope

YukarinnAM, 2022-04-21:
Hello!
After a while using this app with all that it comes for free and the bundle pack, one really starts to need some measure tools in this case i wanted to request for a frecuenciometer
since sometimes we are doing circuits like LC Circuit,A Stable multivibrator or trying to make a Electronic equivalent of a Quartz Crystal , we need to know the exact Frencuency it is, or at least a rought Average
also a being able to measure Power on osciloscope would also make some calcs easier ,even tho this function is not implemented irl osciloscopes, for this as a circuit simulation can make some of our lifes easier when it comes to calculate Avarage Power Consumption on certain components
(Also something more i just thought as i write this is being able to have Amperage and Voltage measures on the osciloscope at the same time)

Once again, Thanks for your time reading this Proto team
and i hope you all have a great day

Proto-App, 2022-05-01:
Hi @YukarinnAM 
Thank you for next great ideas, power should be easy, we would try to add it in next release. Of course frequency meter is on our TODO list but it is harder than power monitoring. Voltage and amperage on one scope is possible but it is relay tricky to do it right (mess on scope) and we postponed this feature.

YukarinnAM, 2022-05-02:
That's great!
Thanks again for the work and the great app

YukarinnAM, 2022-05-03:
also one more thing that i thought but forgot to add was the uptime percentage
it would kinda work only in PWD and maybe would behave weirdly with other ac signals
 also you can easily visualise that a certain average on the oscilloscope so is not really needed but just wanted to add that in case it can result usefull in anything else

Proto-App, 2022-07-05:
Power implemented in version 1.9.1