Proto-App/Proto-Android #142: Issue with themes and exportations

MontMontgomery, 2022-05-06:
Hello!
The new themes are so good, but the 'light' and 'solarized' are a bit tricky to see buttons and switches 
[image]
Here you can see the switch from a relay.
Also a dark-green color in wires, and a bigger outline for components should be a little easier to see.

And about export to pdf, png, jpeg.
It should export the whole circuit, not just the fraction that you are seeing in the moment of the export.

Keep the good work! Great app!

Proto-App, 2022-05-06:
@MontMontgomery  thank you for report, we will improve it.

Proto-App, 2022-06-13:
@MontMontgomery 
Fixed in version 1.9.1

closed 2022-06-13 by Proto-App

MontMontgomery, 2022-06-13:
Neat!