Proto-App/Proto-Android #148: Make Open Source

electron271, 2022-06-12:
I understand that it's hard to make the app open source, but you can still get money from the libraries by having some parts not open source. You can also maybe add a custom plug-in or library interface that you can pay to use.

electron271, 2022-06-12:
I really want to buy some of the libraries and addons, but I would feel better if it was open source. You could also add donations as another source of money, I would definitely donate if it was open-source

Proto-App, 2022-06-14:
@electron271 
Hi, we don't have a plan to open source our application. Donations are agains Google developer policy.

closed 2022-06-14 by Proto-App

electron271, 2022-06-14:
Alright, that sucks that google doesn't let you do that.. I still really like Proto and use it to experiment around without risking blowing my breadboard up